dj-nitehawk/Hybrid-Inverter-Monitor #1: [FR] Configuration file

Saentist, 2022-07-16:
It will be very easy for regular user to have configuration file, 
where can be set some basic settings.
ex.
```
web_port : 8080
dev : hiddev1
```

dj-nitehawk, 2022-07-16:
yes for sure. will add that in the next release. thanks!

dj-nitehawk, 2022-07-16:
btw, have you tried the app with your inverter? does it work? what brand/model inverter do you have?

Saentist, 2022-07-16:
Inverter is mapped to **hiddev1** 
ups remap and take hiddev0

```bash
/dev/usb# ll
total 0
drwxr-xr-x  2 root root    100 юли 16 15:47 ./
drwxr-xr-x 24 root root   4440 юли 16 18:28 ../
crw-------  1 root root 180, 0 юли 16 15:47 hiddev0
crw-------  1 root root 180, 1 юли 11 07:50 hiddev1
lrwxrwxrwx  1 root root      7 юли 16 15:47 seups0 -> hiddev0
```

Inverter is [EASun smv iii 5kva](https://easunpower.com/products/easun-power-usb-bluetooth-5000w-inverter-500vdc-pv-input-230vac-48v-80a-all-in-one-mppt-solar-charger-support-mobile-monitoring-lcd-control) with have ton of clones, Voltronic protocol
Panels are in a shadow from few hours

dj-nitehawk, 2022-07-17:
`v1.2` released with ability to specify web port & usb device path via `appsettings.json` file.
and also you can now customize sunlight period in settings tab.

closed 2022-07-17 by dj-nitehawk

Saentist, 2022-07-17:
> `v1.2` released with ability to specify web port & usb device path via `appsettings.json` file. and also you can now customize sunlight period in settings tab.

Update this info also in [README.md](https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor#readme)