dj-nitehawk/Hybrid-Inverter-Monitor #2: Build documentation

Saentist, 2022-07-17:
Is it possible to be added simple steps for building binary 
needed dependencies etc.

I try to build it on  AlpinLinux 32bit without success.

dj-nitehawk, 2022-07-18:
alpine deps for .net:
https://docs.microsoft.com/en-us/dotnet/core/install/linux-alpine

after installing the deps and then .net sdk, try to publish the `InverterMon.Server` project with following:
```
dotnet publish -r linux-musl-arm --self-contained
```
can't help you anymore than that as i don't have a 32bit alpine device to try stuff on.

closed 2022-07-18 by dj-nitehawk

dj-nitehawk, 2022-07-18:
i added an automated build pipeline which produces for a couple of common linux architectures.
try them out to see if it works with your arch: https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/releases/tag/v1.2

dj-nitehawk, 2022-07-18:
aparrantly `dotnet publish` does not support building for linux-x86 out of the box.
didn't find much clear info on how to do it for 32bit alpine.
most likely you'd have to build the whole .net sdk yourself.
will keep you posted if i find the proper way to do it.