dj-nitehawk/Hybrid-Inverter-Monitor #3: Direct COM port comunication

Saentist, 2022-07-18:
In some cases **hidraw[n]** is not accessible
is there a chance to be implemented  **tty** interface
https://docs.microsoft.com/en-us/dotnet/api/system.io.ports.serialport?view=dotnet-plat-ext-6.0

```bash
# dmesg -T |grep raw
[su jul 17 11:23:59 2022] hid-generic 0003:0665:5161.000C: hiddev2,hidraw2: USB HID v1.11 Device [STMicroelectronics HID in FS Mode] on usb-0000:00:1d.2-1/input0
```
it appear as ttyUSB0

dj-nitehawk, 2022-07-18:
will look in to it when i get some free time.
if you find a good cross-platform library that works, do let me know.
better yet, send in a PR if you got time to spare.

Saentist, 2022-07-19:
https://github.com/genielabs/serialport-lib-dotnet
https://nugetmusthaves.com/Tag/SerialPort

lasitha-sparrow, 2022-10-02:
hi im also having same issue it didnt detect my device as hidraw[n] only as ttyUSB0

[   24.599220] usbcore: registered new interface driver usbserial_generic
[   24.599381] usbserial: USB Serial support registered for generic
[   24.988223] usbcore: registered new interface driver ch341
[   24.988408] usbserial: USB Serial support registered for ch341-uart
[   24.988594] ch341 1-1.3:1.0: ch341-uart converter detected
[   25.012096] usb 1-1.3: ch341-uart converter now attached to ttyUSB0

any idea how to solve this?

dj-nitehawk, 2022-10-02:
@lasitha-sparrow can you run `lsusb` and see if a device like the following is listed:
```
Bus 005 Device 002: ID 0665:5161 Cypress Semiconductor USB to Serial
```
if not, can you post the output of `lsusb` command pls.

lasitha-sparrow, 2022-10-02:
seems like i dont have cypress chip

hi@raspberrypi:~ $ sudo nano /lib/udev/rules.d/99-systemd.rules
hi@raspberrypi:~ $ sudo apt-get ser2net
E: Invalid operation ser2net
hi@raspberrypi:~ $ ^C
hi@raspberrypi:~ $ ^C
hi@raspberrypi:~ $ lsusb
Bus 001 Device 004: ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter
Bus 001 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 001 Device 002: ID 0a05:7211 Unknown Manufacturer hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
hi@raspberrypi:~ $ usb-devicesusb-devices|awk '/1a86/' RS="\n\n"
-bash: usb-devicesusb-devices: command not found
hi@raspberrypi:~ $ usb-devices|awk '/1a86/' RS="\n\n"
T:  Bus=01 Lev=02 Prnt=02 Port=02 Cnt=01 Dev#=  3 Spd=12  MxCh= 0
D:  Ver= 1.10 Cls=ff(vend.) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=1a86 ProdID=7523 Rev=02.64
S:  Product=USB Serial
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=98mA
I:  If#=0x0 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=01 Prot=02 Driver=ch341
hi@raspberrypi:~ $ sudo chmod 777 /dev/ttyUSB0
hi@raspberrypi:~ $ ls -l /dev/ttyUSB0
crwxrwxrwx 1 root dialout 188, 0 Oct  2 01:17 /dev/ttyUSB0

dj-nitehawk, 2022-10-02:
yeah seems like you got a
```
Bus 001 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
```
unfortunately i don't have this device to test with.
somebody who has this device would have to send a PR i'm afraid.

Saentist, 2022-10-02:
```shell
[1710830.730259] usb 3-2: new full-speed USB device number 2 using uhci_hcd
[1710830.918300] usb 3-2: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.54
[1710830.918319] usb 3-2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[1710830.918327] usb 3-2: Product: USB2.0-Ser!
[1710831.563057] usbcore: registered new interface driver usbserial_generic
[1710831.563143] usbserial: USB Serial support registered for generic
[1710831.605167] usbcore: registered new interface driver ch341
[1710831.609076] usbserial: USB Serial support registered for ch341-uart
[1710831.609173] ch341 3-2:1.0: ch341-uart converter detected
[1710831.626355] usb 3-2: ch341-uart converter now attached to ttyUSB0

```
When used RG45>RS232 cable with comes with inverter there is no problem.
Work perfectly just set in **appsettings.json**
```json
    "DeviceAddress": "/dev/ttyUSB0",
```
tested on Ubuntu 20.04

```shell
# update-pciids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  281k  100  281k    0     0   176k      0  0:00:01  0:00:01 --:--:--  176k
Done.
# lsusb
Bus 004 Device 002: ID 0665:5161 Cypress Semiconductor USB to Serial  
^^ INVERTER Micro USB > USB cable
Bus 003 Device 002: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter 
^^ USB to Serial adapter for RG45 to RS232 cable
```

jkwim, 2022-11-29:
Ryan

> yeah seems like you got a
> 
> ```
> Bus 001 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
> ```
> 
> unfortunately i don't have this device to test with. somebody who has this device would have to send a PR i'm afraid.

What if you change the hardcoded device name in:

https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/blob/main/src/Server/InverterService/CommandExecutor.cs

`
 20       var dev = confing["LaunchSettings:DeviceAddress"] ?? "/dev/hidraw0";
        `
        
        Does your code depend on low level drivers or OS level device drivers?

dj-nitehawk, 2022-12-02:
@jkwim `hidraw` is a standard linux driver which mounts **HID** devices under the path `/dev/hidraw{n}` which is simply a file anybody can write bytes to (in correct protocol) in order to read data from attached device (inverter).

`HL-340 USB-Serial adapter` on the other hand is a different kind of device which requires the use of a serial communication library to work (afaik).

so just changing the device address in this case will not work because they require completely different methods of communication.

if i had access to one of these comm boards, i could most likely figure out how to do it. but haven't yet been able to get my hands on one that is compatible with my inverter.

Saentist, 2022-12-02:
@dj-nitehawk 
it's time for auto detection feature ;)
[Matching /dev/hidraw* devices with physical devices](https://arvchristos.github.io/post/matching-dev-hidraw-devices-with-physical-devices/)
Same for com port's 

there is a non open-source software written on dotnet 3.5 with do that.

vjlb, 2023-04-12:
@dj-nitehawk 

Is this what you need to make RS232 Serial work with inverters?

https://github.com/Interster/AxpertInverterLogScriptPython
and this:
https://github.com/jvandervyver/libvoltronic

Hope you make it work, most people are using RS232 Serial to communicate instead USB

dj-nitehawk, 2023-04-12:
added "potential" support for direct serial port communication in [v1.15](https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/releases/tag/v1.15). don't have an actual inverter with a serial interface to test with. so, if you guys have one, do try it out. basically now, if your device is mounted as a HID device, it will be automatically detected by vendor/device ids.

if your inverter is mounted as a serial device at `/dev/ttyUSB{n}`, then update the `appsettings.json` file like so:
```cs
"DeviceAddress": "/dev/ttyUSB0"
```
relaunch the `InverterMon.Server` and it should theoretically work.

closed 2023-04-12 by dj-nitehawk

vjlb, 2023-04-12:
@dj-nitehawk 

Thank you for your effort

Never tested before this latest release
My inverter is on /dev/ttyUSB0
My JK BMS is on /dev/ttyUSB1

Any hint on this fail:

pi@raspberrypi:~/monitor $ sudo ./InverterMon.Server
info: FastEndpoints.StartupTimer[0]
      Registered 7 endpoints in 3547 milliseconds.
info: InverterMon.Server.BatteryService.JkBms[0]
      bms port opened!
info: InverterMon.Server.InverterService.CommandExecutor[0]
      connecting to the inverter...
info: InverterMon.Server.InverterService.CommandExecutor[0]
      inverter connected!
info: Microsoft.Hosting.Lifetime[14]
      Now listening on: http://0.0.0.0:80
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: /home/pi/monitor
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      command error: Read timed out.
info: InverterMon.Server.InverterService.CommandExecutor[0]
      exiting...
pi@raspberrypi:~/monitor $

vjlb, 2023-04-13:
@dj-nitehawk 
Comunication with inverter works ok on this one:
https://github.com/njfaria/SolPipLog
And it shows connected
[image]

Is there a way it can work with JK BMS only ignoring inverter, so i can at least test BMS comunication?

Saentist, 2023-04-13:
> @Saentist will you be able to post a picture of your rs232 pin configuration? i.e. which wire goes to which pins on both ends.

Original console cable from invertor package is what I use, not DIY cable.

[image]

vjlb, 2023-04-13:
I use the same connection

dj-nitehawk, 2023-04-13:
i have the same cable but i don't have a computer with a serial/com port to plug it in to. so i've ordered a usb->ttl adapter so i can cut this cable and make a rj-45 to usb cable myself. just wanted to confirm  what the pin connections are.

pin 1 on rj45 is TX and pin 2 is RX and pin 8 is GND if i'm not mistaken.

hopefully it will work...

@vjlb how have you connected the jk bms to the computer? do you use the jk rs485 adapter? 
there is no support for connecting via bluetooth though.

Saentist, 2023-04-13:
> i have the same cable but i don't have a computer with a serial/com port to plug it in to. so i've ordered a usb->ttl adapter so i can cut this cable and make a rj-45 to usb cable myself. just [...]

USB to TTL 
[image]

or 
USB to RS232
[image]

Let's make difference!!!

dj-nitehawk, 2023-04-13:
@vjlb check out `v1.15.1`
should let you navigate to the bms page without quitting now.

vjlb, 2023-04-13:
OH YEAH! :D
[image]
[image]

vjlb, 2023-04-14:
@dj-nitehawk 

Hey mate, maybe my "Read time out" it's protocol related, i know by mpp-solar my inverter is protocol PI30, maybe are you using diferent default protocol?
[image]

Looks like it could exist diferent protocols:
[image]

And protocols could have diferent commands:
[image]

dj-nitehawk, 2023-04-14:
app only uses PI30. the real issue is serial communication since i haven't been able to test it. most likely would be an easy fix once i get my serial cable working. will keep you posted on progress.

dj-nitehawk, 2023-04-15:
guys can you try [v1.17](https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/releases/tag/v1.17)
i rewrote the device access code (with a bit of help from chat-gpt) 😉
hopefully this one does the job for serial comms.
still haven't got the serial cable for myself to test.
just make sure you've set the correct device paths in `appsettings.json` before launching the app.

vjlb, 2023-04-15:
@dj-nitehawk 

Hey mate we have info, some errors and info is working intermitent:
[image]
https://photos.app.goo.gl/ca11eybiBnmDwhrT6

Saentist, 2023-04-15:
@vjlb 
what inverter model you have?

@dj-nitehawk 
I don't have JBD BMS
how to turnoff this feature?
/comment or delete from config file/

vjlb, 2023-04-15:
@Saentist 

[image]

dj-nitehawk, 2023-04-15:
@vjlb so to confirm, the data shows up on the dashboard but sometimes gets zero'ed out?
if that's the case, it seems like the response from the inverter is not consistent. and sometimes it doesn't accept the `QPIGS` command, which is apparant by the `(NAK` response i see in that log.

let me look in to it a bit more. it's like shooting in the dark without my serial cable :-(

in the meantime, any chance you could try a different computer/cable combo, just to rule out a problem with the physical connection?

@Saentist i'll add a feature to hide/turn off the jk bms section in a future release.

vjlb, 2023-04-15:
@dj-nitehawk 

Yeah, seems like something was messing up with other system, i just switched from Raspberry 3 B to a fresh install of PI OS Lite and Raspberry PI 2 B, everything looks fine now.

Will keep you updated!
Thanks!

dj-nitehawk, 2023-04-15:
> everything looks fine now

so with this new installation, you don't need to use `TroubleMode: yes`?
or do you still need to use trouble mode?

vjlb, 2023-04-15:
> > everything looks fine now
> 
> so with this new installation, you don't need to use `TroubleMode: yes`? or do you still need to use trouble mode?

Trouble mode is set to default "no"
Application is now "clean", no errors:
[image]

vjlb, 2023-04-15:
Found something:

After i tried to switch SBU mode no longer have inverter data on monitor  neither can access settings, also errors came up:
[image]

dj-nitehawk, 2023-04-16:
@vjlb can you run the following command and post the response you get:
```sh
mpp-solar -p /dev/ttyUSB0 -o raw -c POP02
```

vjlb, 2023-04-16:
@dj-nitehawk 

Problem is only when trying to set SBU mode, maybe some code typo?
After trying to set SBU mode i have no longer "Live Stats" all goes to 0 (zero) and application pops errors:

[image]

Also not sure about this, but changing modes(SOL, SBU, UTI) takes longer than mppsolar command or need to press button twice. Need to double check on this

dj-nitehawk, 2023-04-16:
@vjlb try [v1.18.1](https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/releases/tag/v1.18.1)

if the problem persists, do the following in the terminal to get the full path of `mpp-solar` and edit the `appsettings.json` file to set the correct path for `mpp-solar`:

```sh
which mpp-solar
```
also set troublemode to yes.
```json
"TroubleMode": "yes",
"MppSolarPath": "/usr/local/bin/mpp-solar"
```
this workaround should do the trick until i figure out what's going on.

vjlb, 2023-04-16:
@dj-nitehawk 
Changing modes now work all, but need to navigate to "Live Stats" and go back to Settings or else it Button it will circling forever, after going to live starts and go back mode generally is changed. Before was much faster, matter of seconds and no need to navigate to Live stats but with incovenience SBU mode not working.

These are errors now:
[image]

These are settings:
[image]

dj-nitehawk, 2023-04-16:
@vjlb some improvements were made to the ui handling of setting page when using `trouble mode` in the latest version v1.19. 

since it's using mpp-solar underneath for communication for the troublesome commands, it's always gonna be much slower.

it will have to do until i can find the actual cause. maybe i'll have more luck when my serial cable is ready.

dj-nitehawk, 2023-04-16:
guys i'm gonna lock this conversation since it's getting too long. 
pls create new issues for future stuff.
tc!