dj-nitehawk/Hybrid-Inverter-Monitor #4: Optimization's

Saentist, 2022-07-19:
Is it possible to remove all unused code generated on page loading
analyze say that 1.2 megabyte of 1.5 megabytes is useless
[Lighthouse](https://googlechrome.github.io/lighthouse/treemap/?gzip=1#H4sIAAAAAAAACt19W4/jxnbuXzHmKQHMFuvCKtJPk2wDwXk4SJCd/RQYA0piq+mhRJmkpj3e8H9PkdKMtepbq5oaY+chOUiOp7urWJdV6/Kt29/fdS/Dux/+/m5ofrk049Ts/zZ073549zJN5x82G1XpJ+XKJ/Wktf6h1Jt33797bk919+Zf1Zd9O43zzONuaM9TNg1Nc6zP2b6e6vnH7T5MwP3u+3dTO3VN+O1fl99+91/X33734/W3++Y6qu1P4W/+Njb775774bvbHN/V53P4o3HXD2GK06Xrbv/4sR3PXf35//f7eer2FMYc66n91CxTTnXbLaudPp/n30crOoVR4df//fd3p/rYpHc+NGN/GXbNv36e5jEmd79//+a4bd9P4zTU56ft5bTvmqdje3r6ecTpfOmt+f7d5XQJG7/9zPm8jD8yhq9cTuePh6ddf9y8r0/Tp81Bn7t+eq/mz5abfTtOtx+xX1JaK1fZ79/tXtpuPzSn+/2/Nttzvfv4w+bf9H+ECZjRSpfKpkezo4Qhl1O4qmGsu3CBl675sQlk2C40AJNoVd4dxteTxT+schedpDHF7z8xP/tjuqcnnKcoC1uoaJh1zuQFv5lx2OEsTodrjL9d+tLwc7RjVg9D/Tnr2o/N08RdYFFF0/n8bie7/hSo/jSyY62Pt2Pvhj633dQM/EfhTFVu7obu2+fnJmxlx6/Zx9/1998Ne57fJ/9hWHKR06HPl9Nuphhh3UW8Y0OHn9qOHeniU9aeDlyuif+oi6nGKzq23/7c7CbheuMNu+pucFPvXthxBkhVuftD/th85mlCmfiIlKrocgNHFb6qTLxaZXJurwl6zpNEGWYIDL49pc5M54qQ8Wkv3AvstCijgVn43+ZXdngVU0SlyeBhnLJPdXeRtpl+Qc9dPU0NT8VVfLVliSOzfdOchU/HDCg8i7sJDs2UDfXpwC9cmzx+CeQpnIMwFm5FJ497HpjV0iuAJRO6Gpr9ReA1s2yj16TIwGP/SWI18UYDdVNKDBKnPR2EQwZRcf/dsR+mbMvzC22A15h70gpC8heercafdCUdJnLFeKfV/UYXIg4P9zkLTEN4SfGKlSOqSlPzT9DEl2Pu33pXjzw5WODi9/Q7TvUwjdlrO/FMqog1K3u/1ua0T4x1MQ0X919ugvLCn5CPX7s3hBqOgqCMN+rvCeEw9JezSEXAR+mdXAdPfeDmApuogENVJp6AH+ktCB9tq4jFvAZtLds2L/Wntue1DBsfNtHR3h6vFCwjv6ev0+W4bQa96ztxAuSU92d4roexCbxy3154YaoVMEvC5XddfeTPEE6Q8J7n9tdmn23rUdIJ4e6LWM9Z9s4/yojL7Ztde6x5pg7y3kTfCU+CZzpvjmxPU3MQlgicLh58ag6L2beS6+Tc4WTBWBa2zShJOlpBv+dZHl5sNPDcj+03Lv1Y/yqxA6WBJMiKgyX6zUN7fqfA89z9WgPn2TeHYIDzpxTzaG3p2BR1KE3/dn6gtaBKxWu05Jn1Q/gQL0TuxVstmFfRH4ny856nXKXtWo1Eq+gTssZpwAyOjimlzSg4p3vVoutfw3PZSeyI2FjXv13U45WS8n6Z42U7Tu10mSSdDTg+eV6X8zm90PiIKhgsr1yV8fX4WOaJdm0VWYT1cLgcm9MkEFb8IePo8G3fd41A8gbgh8i+2tfC6QLTM9Gqm2EQJGma+GazfQZ6hFsBNaCM2bVgcph4pI5Gnod+6mWwQUVyYWgOWfMrL7DflGeXYEmGXTZrNeHo203XzASxUvG/52I30DerT+2Mhfan7HmoBYXTVLCN8p6L7QJZDetm0rFNYJQnAoc3q3GcIjp2fTmI56AcMMhCk9X3J8GqLWIC1YTD7Zttf5HgLOVQ3bp/+cdgZLa/CR/2oGJ6Rz7cnDPxrAiNNb8Gs1+gLg0cQxGWMaMcwcITtMmYNlUMujXH8yQZs7FCKCpV3sfP1WvHDJatMxuTDpHAoqGTw/5yqlWds6RURswr4vsCdyFGq/RXxG46t7uPwp+BQkosleklcLqpExgsIDOmuif+oBvI6CaAm8X9K5+xikuTtTxZwrshY099L1wYGKZEmOwDe2glcoyPiShQo/RKASl0hCnWuxeBCZYxaVG2Ep4s/xRAj/GEpur9frGsZhx1DK9eUIS1AzXaEfNzaIKsz/b9UVg92Cou1mjG6bNAVYtfiF4TQXLn0S9Ne3jhj6CM33IZf7oPWuCQmkIjaA/rf233AjMpY+IsPbsAeQbt4fgJ7HMFWoMc5WkAtEnClPp9+/w52408UwK6K23M7iVbDwBTbQmO2LWCDHWgPhcEWRPhCm1KFIJEcD/Xu6kXoLVKx1SqK6K2T4JLxwKYZwiN7frTONWCqmFzOGLCqffNOQhjUYPXBmSVI2rO7Mx+27jdXbbtTsLg4EwdgfnP9fSS6azejn0n2lMKwFKliOP13HefRYowHh5gZQnjnI4z6j/0x7CSbfNbKzAyC5LCEhX3CsQtO0pZsQadATbSMOb9HMLFLwBDmFRwMVWgtlUENgt6yn5+2sLl6BJEGEXobpczNgfZClQlIrQlmWVXnz4J2ISpKlAJqXBIXEcJHgpfuYix1cMuXEgwD4TF5zp+fRXxAs9WmTBUAYBPEMv54gLD+JOj+WdLlLE+/HGgky80I2BA8DniCJqPan4/b9CbBt1Yk5sOpy2wAThm5Qh613RdexYYc1XGg0vi5Pzj2Qk2bfxINPH4pak7L2OlqyLxLwdeaVFWg96hjfv9p7uhbHiH1hhzY3WQQ/qB8A7jAnuMeZWuTOlKIeIlSCcODvKgV7ty0SalSfiHqsEidsbHjEaw+jRKZUvv/1gHZssrEhrNIp2XNDJFcrukVb+3GEuF918ShSAw5lfBSlaIcBGpvujcAqwP2CGx4ydJ1VeMeZ27+1Pu6m0jeV8AY6GQ41CfXwQZ5AAccxQQfmkEZGfWzQAtIdIvqE2SQmEQoL1nh3XXHgQnta3AxLUECQw2as9frMIgKRX4y/1uW2GzDkBdSzSwbTvzbMl/CFirIis+ipwzh4vNCR0uUI8AJSoPa549svcvwCwypxZWbQ0ofYYgIous+iS5BAFOMcTrOitoSwDQsRVkHVogRGs8D80kmPcWYLzwk/uNC5ZuUMiAQDThOuOuluzcEoOeFLHy619bgVdpjNIKP8IXLOhwBvy8Rtv7i3ppurOkw4EGW5I4nl3fD/v2JD3koH6AcDckDmm77XnR4HLQfoMpdm+aTH3fTS0P/jgPd+wcARbO9X4vmQAG3a/kZUxDfRrnoGGB38JoclvHevgo+SQNKmKKoGyz0sk/COPAZUMs8sMg4GuVjh9ypWmY5tDIwZIW4J/wk3uyniYBdkKLnMBsXXOQgGoPwjD8JNY4+Y/mGFHnq3uq6p+fx93QCPEICpQ/KiHGTzxFuSq+HEevdYlAF3QkMI9JqOWLwCA9AKDhJ0TBXfQUZmAFMtRVuaBULqwnkzWeIMqA7+UEZroilHMEct91wrsoHCypyAn28/rSCOpPBTqXD+/i/hzq7RwUvmOOwtgcDOBgZM2gG3cYInZlSsD4iDM26bZDyeFcwURvS2eXQ2RGMFgIhCUDAEFQgJZQ5lT2SAFeBkM5CWcYX2rJpVoCUwn6L5U7x3N/ko4sMD5QPZ2y8aqz9Cw+zwGqpFbxy3Ts3pikKEAaBMWpovS3OEobzqGAMrCQEglm5VTE2HSOQWq+Wof2BuUIhLhXKtrCqZ9qPgdD5TkTizwTJbcNcQvKgcBQFCVJWE9wCSonjjUJGUHZ6EmkydAcJMHoAPotiJAKdy4ETyPnLkhQ1JyIlCW0CJMDy9MeZ0isXeVgM1a+gH1nieSPErbvA9MiNBMUXg5NAn9MWSiB5kWG6xA7DQxklYNCG4QGKgJo7NphJ6n5Gm3I3BZk37uhHwPvaweGW3jQE6yvhKwfObDTgfKrLMFzZO8MHhvxrqe2noMjK4jemFsmAAOire7GMVtcKyvhWeIWmb8k6FKMTNJUK5r1ZM5LCPkQWs122WM3YwGwU1rn6w54IYQY7FtzqQbU5dm/fL/nq67NiQ2004MQLx6myAr4iaLm2C4IwIPovCsgIysYppEXbmpPl16IsA7iBvykXlN2dDMnWTRqfk/chtNUWtC4yzfIv1JuDQmHZwYGQWWCERLnKi4/vLdZ2RtWS6Ihu7eUY7OkfjUpDwLAeUfixJY18TcOoYBlrmGLypJ1jF27bxiuWloMFy28oH8kt61gIqoP1nO+roSXFZgyR5LtrsuXdAk4SVNR+g1WazCgtvWw+fpfgj0AoZLWEvx6ancfs2MzvfQMwRgLeZlGl4J8ntWEUdIQkCs5yhAECB3Ct4g5/MslXF4rsFADuR4GAN4sgRwGRRJRPL8yaIKA38uXpD2CsUvUjwQoqxFyJwQ6XDcnsFl0bGgaJC/B3xrweq1p5DePiugCjGti5Z0lpB78nIqE7LyBXINJrwIb+YkRRtyXLapW+mFxmIPDrSSmccLRgGqtpmv/Kgm5pw/OOrt44B5bvUIBpFcSKOKbmnh2JEqpwNngCYgkUUoBoXyWpFaJB60wNbOyGviMEAlYYlgW0fKSTEqjqR5ZEF+CFzcytzFgcmsSGdacDuGWEhMwAauEzy5ANOcQthhebXLJPS7D2YGD4DMlDpNE0jVQOQmaStgwjPtQ557QtpjXgXmERU5xf9EPrjymilkiV46tmDWhSnDSqsLT/GUpYESBnhV4YaSiSeHAYO6Zeb+ETmcPYL3jYaEKl13mTlK0A1eT0B3tkNYVSQK9Wwe/F4yboUGoQ3081kP21jzOKsj8taVeRbMavHQ6p69+sR1SL5Ya0C/1wLzPQM0YceurXBBhMrDvASUxJH7iU9tIwrtEf2bhiOt4WbxAsZhFXuWWAkv7ny8jez9Y2aMQC9NcZ5FcMvhsDHUh7wVwL9j7mC5MntzP7SSqzBCMaqn7aaql0H+N/Jz6YcbPx6DzD0IgSIlpF5Y+9kPTz+MZpSnsGH19uSqFt55IWM5R8TOOhtuJkUoKUGyVGxod19T8qWMQr7+XCY141xBZr4hC/dLUk5RvYiygwLqimWmp+IoSCnB4gi/IMHsBR0yk2BLPKAQawCWTsIpUAKSDkyoIWjvuAvPjb8cyBQBcpBLfPDSbJUhKCJOwENxhNAkfXvxVrJsLqdtW6lFssCgw5pSitgnFpYJoGOoRT8VvKo11U3K1zlWhqwpDTku/htCCmsvE89OUqkSgBNB3eM2rGIE1sF3jSVBbilBLFR+006sj3We/K9IqMdBFTQ2BWJ0TzifqR6YETdoootSKqT2g0moT+TLm+B/mOXs440AVjzpxtMUsz4rYybugJjSjlLZfYFQeiSx+aTpJH7foxjGGQhOdiMFgyEBeRR7fKQja7WX+JaOMAmpqKvOokW4cIIZGkfIIcmQthpEr4iruz/VOSp1ToN14+jzGVlSfDXJgkkUiBwxoJh14VcqeAjzCUzhljk/jQMhSCAFJuNEJEicGYxvwOAd6ijyvIqQKHiLCm0TmAudO1JuU2IHowFjgy04pBfYO8Q2kmDBX6MuuYYWWgYPUGqGhSthq0PjVKjmHIKd2a0UzRKuFhwhFKS2JhNo3z6x2AU/EOiVwlcNcC0WO6GKyvXwU2LMX8DjMiPKGxmGI0dk5BJJVtIZlJ8SCGhDZmqgZ57kS3iCIELA2LfE7pnIkGTsv15HJNE6XgRMC2gHVqCWFkb2u6zSC7xfc1hVVkwRIBKtiUEYe/p5XpB0GPJG6JeNrK3BxB8RR6HW+FbQ87rmBYGFZVIs09QPMTHrc7LpeMqqVfjNjSDlVYDiGMa4QcAc2YyiQjgP7tcyrOT+UBS9+bcdN/6kZOq7YrcZKPdcfsVNdpj67MikhQP2tDOplhpd2L4lgcCcoGuG6jB96MV3EwUubdXIBiNxIeOQc8wQVDlX1qOJVeMDFyMPZ1bNrViAo6+BeiAmdFP4e0kHo8xFVDpeDcAtWCWHLu7nWWDJijIG/PZGtu8swihX0gNv43DDg51JHIBtfGs71UBWSzd2Jyf/WE4tgHyxO6e/oegR41VYVuH5s+B85xmIalrBnjhYsWItW6UqwomTmiJljNHImmcHBe0Tuxaf4EipI7HQEC/8jZpQnZwPKmitIyVExY8Zb5Aie6LSJ5AKnwfNXOI1xJoLpBM47bSpa0U7OWlHXOPvYiWQ96gwChgX6YqAXCoZ19ef+wjruQE2SPdu35JlsJxXpAYHsbL5Kkmtgn5r4wWZZIIS1APfLK5DmmyVDKzuHSxBiAzDAw5ucC9fefK1vxbEBOEzrvSCrE4eBPJVELcsZvZDAVNLCwks24rUUwuUslnJTFpiZoiUVfusF6MhBJIojHvDnWlAHLLAra2kaU/8qlVvkKgzff/R6+cJgh7nUJKj0tf4kjtUYy1RQ0PwKeyQPG0JTqxjPFPcNhz3X3lzpnMkLtKZ9aQTt9JapknBDBjEMNXEsgWYXFJ4hWnzCQQYIWmnC8AIOQtxcorgK8haSLWiAXtrhAqzXa0oCEmCoPHqmLJEZ51aMxIG4WEdNmAUz28guNY3ZUcQ45isPKIM6+/IjoTvFsd83XRZ0JqlePBOTRU9gqSHx3Dbdnhf5DN5Mmcbl2sBFeL5ptXs5xEwEEUu8BTXD5LHw4YQuUxTQCSZHe5rL1vQfGR9U4B1YHkBqnyK7/iCiQ/lV5UOYSJ2ckr6MA2p49Joksc+gZbJ4NtguuloFeoa3DcLCRlbjIiE5jRprWEma0tfyLyK0iBl2eXx6Yt0YrEViTUR5Vw5w9aVxDADcp+Hzgrkyfj5ue8F/6kDlK2jXFJFhO9ByrIufD4esuLwED2Kh/UONdgqbQ5lc6xYPDo+0iWLUGdAftBWDSwRfsjYO8TpVWGFLPFMJigxUDVv8WbwmL/U/sNBWoKDtj+aiRItWstC41DsI7tbHhY3k4oDMOyNOlPHctVPq8wq9xJEyP/MlFtHCyGQJIBO1cI8djHSx6kmoHICgingbxklo21JCmM+c+UfMl6GpOTaMxcZKKc9ZRpLeSHNPJZkCbFasS9eDDjua1MUZj32/un4KjZG9Ds1kFxZG25cUEJ4jgzjqwvzMxXJ/TNE1TOFIH6kcwr4hYibXa+Q8U+M/p9h+4rCxHDmBBz+9fFrLQaIYhC9qOfdN4MlmEbePPWMIcfHVuphOptwTyYh8vpxOUtEmbO9D3KMvfdcFQ1jUbgxEYRsVx+gLOAril8Tx83moj0IlEYMNGUqaXiPg3h6jahx9SQuTZ56SxrI6izvgoRtW2KmD4iUJiAZc7orAddd7EtRmptbLqqgrC85Bq2jfqE64IS5/KY8OelE3OU0R4lKu5/YY1/IAKTnKr3+51AK8WYCdNcfH0Ry4OUyQ47iyUSWy1/vnJlUpUqj2qoKg3LslJ1DS/DHHpAL/vi6MiiDcbdNtJCA3yDB4hUUlqH83F6EQVAMouqYN1ZbqXLNxkyjl8Uaq1tcpUlVIYUNzobGfHvbwae+w1KxWVpC8N6iLYxdYabZcAkRYVX/2W8pdmApwfRnSFiBVJQ8DMXLia7y67SR/HY6l8uFjtv2cJdpeWeyMqEnt4KU9YJbYQIXZ8I6GQy4z/IltjO3p0KXWYJl2JRYn+BNLeGkPL4sDMpMVMlRTnOKmEIKwAMIz7Oj0fWKZOE1KnZBpBE8jPHZL4rKu15neDVMBrWRu9I1J4FY16dJyo8zwwneTVB8RpR3Bpf7c4BtZ/pk5ErVXPDbVIq0Ixnbbze67xBQVgPjeKz6qnjn+CjKyjMhm54y6xAPVTMk1n9NUhrltzZ9hdWF8mp6C5QMp2SRDjM4hv3SFDjWfx/P8KbJYZricgiK0+yhO8ZYSeKxHpiO3KSF1LBiOj+Y5aMQ2lFcr8xwMpqgbWpdIsIc4U4xURGqPsjFVYAtsRzs8SkYv1osglRZuCIOMDWPpNhrYdOtJD/4M9EcqCQFMMAGm6h7F1hc2muoA7kDDVTQBewUrYtJTCnoKc64h5xKCUGpbSYnI28s0SaFNFpskaCKf90MtlX0EAJNESMtthXMAxUti/19d2elCpA6jnbFa7XWOTtTRsDb4XOgJZhEDATzgitfgvBjf37D4vCsxiMhK1cDQBtj8kfollvfROQa+FKUUGX1zAAhxdFix1iOt80SOYABR5b8EYGaJIncAR1Y2jje+rn5zb0Fuzi2DUdoSes3ZUgJW9u14y2cRhB56dz1R6pYeOYLtx8icJfP0j/93R0Yc71ZYw9KGs5KKWJ36ffPh2O8vXSNEpWGOt5dyWT9M [... cut ...]
http://192.168.1.222:82:1:0 | Unknown error occurred while trying to verify integrity.
http://192.168.1.222:82:1:0 | Unknown error occurred while trying to verify integrity.
http://192.168.1.222:82:1:0 | Unknown error occurred while trying to verify integrity.
http://192.168.1.222:82:1:0 | Unknown error occurred while trying to verify integrity.
http://192.168.1.222:82:1:0 | Unknown error occurred while trying to verify integrity.
http://192.168.1.222:82:1:0 | Unknown error occurred while trying to verify integrity.

```

dj-nitehawk, 2022-07-19:
does treeshaking exist for blazor wasm?
don't think it does.

dj-nitehawk, 2022-07-19:
about the network errors, it looks like browser blocked some resources.
was this built on alpine 32bit?
maybe it wasn't built correctly.

closed 2022-07-19 by dj-nitehawk

Saentist, 2022-07-19:
> about the network errors, it looks like browser blocked some resources. was this built on alpine 32bit? maybe it wasn't built correctly.

any browser same issue
even on phone
**linux-x64** from releases
Ubuntu 20.04 x64
**linux-musl-x64.zip**
do not work

dj-nitehawk, 2022-07-19:
try v1.2.1
works fine here
https://streamable.com/i4s9g4