dj-nitehawk/Hybrid-Inverter-Monitor #5: BUG Input string was not in a correct format.

Saentist, 2022-07-19:
Not idea what this error is 
browser show only this

[image]

```bash
./InverterMon.Server
info: FastEndpoints.StartupTimer[0]
      Registered 6 endpoints in 6549 milliseconds.
info: InverterMon.Server.InverterService.CommandExecutor[0]
      inverter connected!
info: Microsoft.Hosting.Lifetime[14]
      Now listening on: http://0.0.0.0:82
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: /opt/INV/INV-MON/
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
^Cinfo: Microsoft.Hosting.Lifetime[0]
      Application is shutting down...
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.

```

Saentist, 2022-07-19:
I get this from [MPQTT](https://github.com/lluiscab/MPQTT) software with use same hidraw (not in same time)
/in a moment sun is 90 degrees from panels front/
```
 DEBUG masterpower_api::codec > Encoding command QuerySerialNumber.
 TRACE masterpower_api::codec > Command payload (QuerySerialNumber): ().
 TRACE masterpower_api::codec > Encoded command (QuerySerialNumber): [81, 73, 68, 214, 234, 13]
 TRACE masterpower_api::inverter > Writing command to stream
 DEBUG masterpower_api::codec    > Decoding response QuerySerialNumber.
 TRACE masterpower_api::codec    > Decoding response (QuerySerialNumber): [40, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 186, 81].
 ERROR mpqtt                     > ParseInt(ParseIntError { kind: InvalidDigit })
```

dj-nitehawk, 2022-07-19:
so even [MPQTT](https://github.com/lluiscab/MPQTT) has a problem parsing what the inverter returns.
best way to move forward is to update the source code like i mentioned and see what the inverter returns.
and adjust the code accordingly.

does your inverter work with [watchpower](https://watchpower.software.informer.com/)
if watchpower works, then there's hope.

closed 2022-07-19 by dj-nitehawk

Saentist, 2022-07-19:
after some power on panel
```bash
 DEBUG masterpower_api::codec > Encoding command QuerySerialNumber.
 TRACE masterpower_api::codec > Command payload (QuerySerialNumber): ().
 TRACE masterpower_api::codec > Encoded command (QuerySerialNumber): [81, 73, 68, 214, 234, 13]
 TRACE masterpower_api::inverter > Writing command to stream
 DEBUG masterpower_api::codec    > Decoding response QuerySerialNumber.
 TRACE masterpower_api::codec    > Decoding response (QuerySerialNumber): [40, 53, 53, 51, 53, 53, 53, 51, 53, 53, 53, 51, 53, 53, 53, 50, 253].
 TRACE masterpower_api::codec    > Decoded response (QuerySerialNumber): QIDResponse { serial_number: 55355535553555 }
 DEBUG masterpower_api::codec    > Encoding command QueryProtocolId.
 TRACE masterpower_api::codec    > Command payload (QueryProtocolId): ().
 TRACE masterpower_api::codec    > Encoded command (QueryProtocolId): [81, 80, 73, 190, 172, 13]
 TRACE masterpower_api::inverter > Writing command to stream
 TRACE masterpower_api::codec    > Invalid response format (QueryProtocolId): [0, 0, 0, 0, 0, 0, 40, 80, 73, 51, 48, 154, 11]
 TRACE masterpower_api::codec    > Attempting to recover from invalid response
 DEBUG masterpower_api::codec    > Decoding response QueryProtocolId.
 TRACE masterpower_api::codec    > Decoding response (QueryProtocolId): [40, 80, 73, 51, 48, 154, 11].
 TRACE masterpower_api::codec    > Decoded response (QueryProtocolId): QPIResponse { protocol_id: 30 }
 DEBUG masterpower_api::codec    > Encoding command QueryFirmwareVersion.
 TRACE masterpower_api::codec    > Command payload (QueryFirmwareVersion): ().
 TRACE masterpower_api::codec    > Encoded command (QueryFirmwareVersion): [81, 86, 70, 87, 98, 153, 13]
 TRACE masterpower_api::inverter > Writing command to stream
 DEBUG masterpower_api::codec    > Decoding response QueryFirmwareVersion.
 TRACE masterpower_api::codec    > Decoding response (QueryFirmwareVersion): [40, 86, 69, 82, 70, 87, 58, 48, 48, 48, 52, 49, 46, 49, 53, 220, 170].
 TRACE masterpower_api::codec    > Decoded response (QueryFirmwareVersion): QVFWResponse { major: 65, minor: 21 }
 DEBUG masterpower_api::codec    > Encoding command QueryFirmwareVersion2.
 TRACE masterpower_api::codec    > Command payload (QueryFirmwareVersion2): ().
 TRACE masterpower_api::codec    > Encoded command (QueryFirmwareVersion2): [81, 86, 70, 87, 50, 195, 245, 13]
 TRACE masterpower_api::inverter > Writing command to stream
 TRACE masterpower_api::codec    > Invalid response format (QueryFirmwareVersion2): [0, 0, 0, 0, 0, 0, 40, 78, 65, 75, 115, 115]
 TRACE masterpower_api::codec    > Attempting to recover from invalid response
 DEBUG masterpower_api::codec    > Decoding response QueryFirmwareVersion2.
 TRACE masterpower_api::codec    > Decoding response (QueryFirmwareVersion2): [40, 78, 65, 75, 115, 115].
 ERROR mpqtt                     > InvalidPayload(None)
```
but app still show red messages

```bash
root@ET1610PT:/opt/INV/INV-MON# ll
общо 57132
drwxr-xr-x 3 root root     4096 юли 19 13:45 ./
drwxr-xr-x 5 root root     4096 юли 19 13:41 ../
-rw-r--r-- 1 root root      201 юли 19 13:45 appsettings.json
-rw-r--r-- 1 root root      202 юли 17 20:42 hidrawmatch.sh
-rw-r--r-- 1 root root    57344 юли 19 10:52 InverterMon.db
-rw-r--r-- 1 root root 58420271 юли 19 11:12 InverterMon.Server
drwxr-xr-x 5 root root     4096 юли 19 11:12 wwwroot/
root@ET1610PT:/opt/INV/INV-MON# chmod +x InverterMon.Server
root@ET1610PT:/opt/INV/INV-MON# info: FastEndpoints.StartupTimer[0]
      Registered 5 endpoints in 3515 milliseconds.
info: InverterMon.Server.InverterService.CommandExecutor[0]
      inverter connected!
info: Microsoft.Hosting.Lifetime[14]
      Now listening on: http://0.0.0.0:82
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: /opt/INV/INV-MON/
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
[... 23 lines cut ...]
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
^Cinfo: Microsoft.Hosting.Lifetime[0]
      Application is shutting down...
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.

```
Interface is loading

[image]

Saentist, 2022-07-19:
Just for information, possible solution described here 
https://github.com/ned-kelly/docker-voltronic-homeassistant/issues/20

Saentist, 2022-07-23:
How to see this string?

with QPIGS command i see 
[image]

but still this error

dj-nitehawk, 2022-07-23:
try this [debug build](https://filedn.com/lnBhfXoAIklXFOaQ7BSCnmp/linux-x64.7z)
it will log the response from the inverter along with a stack trace.
let's see what your answers.

Saentist, 2022-07-23:
Change TAG from **CLOSED** to **UNDER-INVESTIGATION** ;)

Saentist, 2022-07-23:
badly panels are in shade
[image]

but response is 
```
./InverterMon.Server
info: FastEndpoints.StartupTimer[0]
      Registered 6 endpoints in 3348 milliseconds.
info: InverterMon.Server.InverterService.CommandExecutor[0]
      inverter connected!
info: Microsoft.Hosting.Lifetime[14]
      Now listening on: http://0.0.0.0:82
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: /opt/INV/INV-MON/
INVERTER RESPONSE: (000.0 00.0 000.0 00.0 0000 0000 000 000 00.00 000 000 0000 00.0 000.0 00.00 00000 00000000 00 00 00000 000??
   at System.Number.ThrowOverflowOrFormatException(ParsingStatus , TypeCode )
   at System.Number.ParseDecimal(ReadOnlySpan`1 , NumberStyles , NumberFormatInfo )
   at System.Decimal.Parse(String s)
   at InverterMon.Server.InverterService.Commands.GetStatus.Parse(String responseFromInverter) in D:\SOURCE-CONTROL\InverterMon\src\Server\InverterService\Commands\GetStatus.cs:line 20
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
```
strange this build stop to add new lines as regular version

dj-nitehawk, 2022-07-23:
send the log when the values are not all zeros. 
and yes I'm stopping execution after one error. 
cause just 1 stack trace is all I need. 
but need some real power values to see what's really going on. 
do you not have grid power connected to the inverter? and also no batteries connected? only solar connected?

Saentist, 2022-07-23:
Yes total offline ;)
funny is that charger will blow battery I use with this 10A minimum, so I charge with external charger.
But am in process of create garden solar array with 4.5KW and rearrange cabling to hybrid output from inverter, as should be

Saentist, 2022-07-24:
So sun return and data come ;)

```
./InverterMon.Server
info: FastEndpoints.StartupTimer[0]
      Registered 6 endpoints in 3613 milliseconds.
info: InverterMon.Server.InverterService.CommandExecutor[0]
      inverter connected!
INVERTER RESPONSE: (000.0 00.0 000.0 00.0 0000 0000 000 237 00.10 000 000 0028 00.0 242.1 00.00 00000 00000000 00 00 00000 010?T
info: Microsoft.Hosting.Lifetime[14]
      Now listening on: http://0.0.0.0:82
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: /opt/INV/INV-MON/
   at System.Number.ThrowOverflowOrFormatException(ParsingStatus , TypeCode )
   at System.Number.ParseDecimal(ReadOnlySpan`1 , NumberStyles , NumberFormatInfo )
   at System.Decimal.Parse(String s)
   at InverterMon.Server.InverterService.Commands.GetStatus.Parse(String responseFromInverter) in D:\SOURCE-CONTROL\InverterMon\src\Server\InverterService\Commands\GetStatus.cs:line 20
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      execution error: Input string was not in a correct format.
```

if this can make your live easyer

```
Parameter                       Value           Unit
ac_input_voltage                0.0             V
ac_input_frequency              0.0             Hz
ac_output_voltage               0.0             V
ac_output_frequency             0.0             Hz
ac_output_apparent_power        0               VA
ac_output_active_power          0               W
ac_output_load                  0               %
bus_voltage                     245             V
battery_voltage                 0.1             V
battery_charging_current        0               A
battery_capacity                0               %
inverter_heat_sink_temperature  28              °C
pv_input_current_for_battery    0.0             A
pv_input_voltage                250.3           V
battery_voltage_from_scc        0.0             V
battery_discharge_current       0               A
is_sbu_priority_version_added   0               bool
is_configuration_changed        0               bool
is_scc_firmware_updated         0               bool
is_load_on                      0               bool
is_battery_voltage_to_steady_while_charging     0               bool
is_charging_on                  0               bool
is_scc_charging_on              0               bool
is_ac_charging_on               0               bool
rsv1                            0               A
rsv2                            0               A
pv_input_power                  0               W
is_charging_to_float            0               bool
is_switched_on                  1               bool
is_reserved                     0               bool
```

dj-nitehawk, 2022-07-24:
can you give v1.4 a try?
https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/releases/tag/v1.4

i believe the parsing issue is caused by a system culture issue.
so i've locked the culture of the app domain to `en-US`
let's see if that does the trick.

Saentist, 2022-07-24:
We have some success but new problem come
[image]
**Menu missing**
/actually I found next BUG hour ago but not expect to have such effect/
http://xxx.xx/api/settings/get-charge-ampere-values **ERROR 404**
https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/blob/760b1df9ab9eb01041bc663cced9beea3068d076/src/Server/InverterService/Commands/GetChargeAmpereValues.cs#L5
inverter do not respond on command QMCHGCR

```
Command: command help - List available commands for protocol PI30
------------------------------------------------------------
Parameter                       Value           Unit
F                               Set Device Output Frequency -- examples: F50 (set output frequency to 50Hz) or F60 (set output frequency to 60Hz)
MCHGC                           Set Max Charging Current (for parallel units) -- examples: MCHGC040 (set unit 0 to max charging current of 40A), MCHGC160 (set unit 1 to max charging current of 60A)
MNCHGC                          Set Utility Max Charging Current (more than 100A) (for 4000/5000) -- example: MNCHGC1120 (set unit 1 utility max charging current to 120A)
MUCHGC                          Set Utility Max Charging Current -- example: MUCHGC130 (set unit 1 utility max charging current to 30A)
PBCV                            Set Battery re-charge voltage -- example PBCV44.0 - set re-charge voltage to 44V (12V unit: 11V/11.3V/11.5V/11.8V/12V/12.3V/12.5V/12.8V, 24V unit: 22V/22.5V/23V/23.5V/24V/24.5V/25V/25.5V, 48V unit: 44V/45V/46V/47V/48V/49V/50V/51V)
PBDV                            Set Battery re-discharge voltage -- example PBDV48.0 - set re-discharge voltage to 48V (12V unit: 00.0V/12V/12.3V/12.5V/12.8V/13V/13.3V/13.5V/13.8V/14V/14.3V/14.5, 24V unit: 00.0V/24V/24.5V/25V/25.5V/26V/26.5V/27V/27.5V/28V/28.5V/29V, 48V unit: 00.0V/48V/49V/50V/51V/52V/53V/54V/55V/56V/57V/58V, 00.0V means battery is full(charging in float mode).)
PBFT                            Set Battery Float Charging Voltage -- example PBFT58.0 - set battery float charging voltage to 58V (48.0 - 58.4V for 48V unit)
PBT                             Set Battery Type -- examples: PBT00 (set battery as AGM), PBT01 (set battery as FLOODED), PBT02 (set battery as USER)
PCP                             Set Device Charger Priority -- examples: PCP00 (set utility first), PCP01 (set solar first), PCP02 (HS only: set solar and utility), PCP03 (set solar only charging)
PCVV                            Set Battery C.V. (constant voltage) charging voltage -- example PCVV48.0 - set charging voltage to 48V (48.0 - 58.4V for 48V unit)      
PE                              Set the enabled state of an Inverter setting -- examples: PEa - enable a (buzzer) [a=buzzer, b=overload bypass, j=power saving, K=LCD go to default after 1min, u=overload restart, v=overtemp restart, x=backlight, y=alarm on primary source interrupt, z=fault code record]
PD                              Set the disabled state of an Inverter setting -- examples: PDa - disable a (buzzer) [a=buzzer, b=overload bypass, j=power saving, K=LCD go to default after 1min, u=overload restart, v=overtemp restart, x=backlight, y=alarm on primary source interrupt, z=fault code record]
PF                              Set Control Parameters to Default Values -- example PF (reset control parameters to defaults)
PGR                             Set Grid Working Range -- examples: PCR00 (set device working range to appliance), PCR01 (set device working range to UPS)
POP                             Set Device Output Source Priority -- examples: POP00 (set utility first), POP01 (set solar first), POP02 (set SBU priority)
POPLG                           Set Device Operation Logic -- examples: POPLG00 (set Auto mode), POPLG01 (set Online mode), POPLG02 (set ECO mode)
POPM                            Set Device Output Mode (for 4000/5000) -- examples: POPM01 (set unit 0 to 1 - parallel output), POPM10 (set unit 1 to 0 - single machine output), POPM02 (set unit 0 to 2 - phase 1 of 3), POPM13 (set unit 1 to 3 - phase 2 of 3), POPM24 (set unit 2 to 4 - phase 3 of 3)
PPCP                            Set Parallel Device Charger Priority (for 4000/5000) -- examples: PPCP000 (set unit 1 to 00 - utility first), PPCP101 (set unit 1 to 01 - solar first), PPCP202 (set unit 2 to 02 - solar and utility), PPCP003 (set unit 0 to 03 - solar only charging)
PPVOKC                          Set PV OK Condition -- examples: PPVOKC0 (as long as one unit has connected PV, parallel system will consider PV OK), PPVOKC1 (only if all inverters have connected PV, parallel system will consider PV OK)
PSDV                            Set Battery Cut-off Voltage -- example PSDV40.0 - set battery cut-off voltage to 40V (40.0 - 48.0V for 48V unit)
PSPB                            Set Solar Power Balance -- examples: PSPB0 (PV input max current will be the max charged current), PSPB1 (PV input max power will be the sum of the max charge power and loads power)
PBATCD                          Battery charge/discharge controlling command -- examples: PBATCDxxx (please read description, use carefully)
DAT                             Set Date Time -- examples: DATYYYYMMDDHHMMSS (14 digits after DAT)
PBATMAXDISC                     Battery max discharge current -- examples: PBATMAXDISCxxx (000- disable or 030-150A)
Q1                              Q1 query
QBOOT                           DSP Has Bootstrap inquiry
[... cut ...]
so solution can be 

```
Command: QMUCHGCR - Max Utility Charging Current Options inquiry
------------------------------------------------------------
Parameter                       Value           Unit
max_utility_charging_current    002             A
unknown_value_in_response_1     010
unknown_value_in_response_2     020
unknown_value_in_response_3     030
unknown_value_in_response_4     040
unknown_value_in_response_5     050
unknown_value_in_response_6     060
```

where 
002 in DEC is equal to
010 in BIN

dj-nitehawk, 2022-07-24:
for missing top menu, just do a CTRL+F5 page refresh.

Saentist, 2022-07-24:
clearing cache helped but Settings still not opens

```shell
./InverterMon.Server
info: FastEndpoints.StartupTimer[0]
      Registered 6 endpoints in 3399 milliseconds.
info: InverterMon.Server.InverterService.CommandExecutor[0]
      inverter connected!
info: Microsoft.Hosting.Lifetime[14]
      Now listening on: http://0.0.0.0:82
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: /opt/INV/INV-MON/
fail: Microsoft.AspNetCore.Server.Kestrel[13]
      Connection id "0HMJDDUK1GG9H", Request id "0HMJDDUK1GG9H:00000002": An unhandled exception was thrown by the application.
      System.Threading.Tasks.TaskCanceledException: A task was canceled.
         at InverterMon.Server.InverterService.Commands.Command`1.WhileProcessing(CancellationToken c)
         at InverterMon.Server.Endpoints.Settings.GetChargeAmpereValues.Endpoint.HandleAsync(CancellationToken c)
         at FastEndpoints.Endpoint`2.ExecAsync(HttpContext ctx, EndpointDefinition endpoint, CancellationToken cancellation)
         at Microsoft.AspNetCore.Authorization.AuthorizationMiddleware.Invoke(HttpContext context)
         at Microsoft.AspNetCore.Server.Kestrel.Core.Internal.Http.HttpProtocol.ProcessRequests[TContext](IHttpApplication`1 application)
fail: Microsoft.AspNetCore.Server.Kestrel[13]
      Connection id "0HMJDDUK1GG9K", Request id "0HMJDDUK1GG9K:00000015": An unhandled exception was thrown by the application.
      System.Threading.Tasks.TaskCanceledException: A task was canceled.
         at InverterMon.Server.InverterService.Commands.Command`1.WhileProcessing(CancellationToken c)
         at InverterMon.Server.Endpoints.Settings.GetSettingValues.Endpoint.HandleAsync(CancellationToken c)
         at FastEndpoints.Endpoint`2.ExecAsync(HttpContext ctx, EndpointDefinition endpoint, CancellationToken cancellation)
         at Microsoft.AspNetCore.Authorization.AuthorizationMiddleware.Invoke(HttpContext context)
         at Microsoft.AspNetCore.Server.Kestrel.Core.Internal.Http.HttpProtocol.ProcessRequests[TContext](IHttpApplication`1 application)
```

dj-nitehawk, 2022-07-24:
does it take more than 5 seconds in inverter_poller to retrieve the result of QMUCHGCR?

dj-nitehawk, 2022-07-24:
let's see the log from this debug build:
https://filedn.com/lnBhfXoAIklXFOaQ7BSCnmp/debug-linux-x64.7z

Saentist, 2022-07-24:
result of QMCHGCR
```JSON
{"error": "NAK"}
```
```
Parameter                       Value           Unit
error                           NAK
```

```shell
./InverterMon.Server
info: FastEndpoints.StartupTimer[0]
      Registered 6 endpoints in 3556 milliseconds.
info: InverterMon.Server.InverterService.CommandExecutor[0]
      inverter connected!
info: Microsoft.Hosting.Lifetime[14]
      Now listening on: http://0.0.0.0:82
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: /opt/INV/INV-MON/
QMCHGCR: (NAKs
QMUCHGCR: (NAKs
QMCHGCR: (NAKs

```

dj-nitehawk, 2022-07-24:
so your inverter doesn't accept QMCHGCR/QMUCHGCR commands.
no idea what to do about that :-(
can you send QMCHGCR/QMUCHGCR with any other applications? they all get a "NAK" from the inverter right?

Saentist, 2022-07-24:
```
Command: QMUCHGCR - Max Utility Charging Current Options inquiry
------------------------------------------------------------
Parameter                       Value           Unit
max_utility_charging_current    002             A
unknown_value_in_response_1     010
unknown_value_in_response_2     020
unknown_value_in_response_3     030
unknown_value_in_response_4     040
unknown_value_in_response_5     050
unknown_value_in_response_6     060

QMCHGCR
Parameter                       Value           Unit
error                           NAK
```

dj-nitehawk, 2022-07-24:
so QMCHGCR command doesn't work on your inverter.

try this build:
https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/releases/tag/v1.4.1

it should hopefully not crap out when the inverter doesn't respond successfully.

Saentist, 2022-07-24:
Settings show Loading... and nothing more
no error in console

dj-nitehawk, 2022-07-24:
in that case there's not much more I can do from here. you will have to debug the source code yourself with your inverter. 

if your find a fix, do send in a PR. 

good luck!

Saentist, 2022-07-24:
Badly I'm not programming.
I can collect information and make suggestions

good suggestion in this case is to be able to run app with some options as

--report
with will create report text file with all commands https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/issues/5#issuecomment-1193306878

Add tags to this project, that will help more people to find it, 
hope some one will have skills to help with code.

Saentist, 2022-07-24:
by the way when reading API commands 
QMCHGCR
can be replaces by
```
Command: QMUCHGCR - Max Utility Charging Current Options inquiry
------------------------------------------------------------
Parameter                       Value           Unit
max_utility_charging_current    002             A
unknown_value_in_response_1     010
unknown_value_in_response_2     020
unknown_value_in_response_3     030
unknown_value_in_response_4     040
unknown_value_in_response_5     050
unknown_value_in_response_6     060
```

**because QMCHGCR its not supported in USB mode only in RS232**

other interesting QUERY commands
QMN (some product identification)
QSID (advanced version of QID)
QBEQI (battery charging parameters)
QET<CRC><cr>: Query total PV generated energy
QEYyyyy<CRC><cr>: Query PV generated energy of year
QEMyyyymm<CRC><cr>: Query PV generated energy of month
QEDyyyymmdd<CRC><cr>: Query PV generated energy of day
QLT<CRC><cr>: Query total output load energy
QLYyyyy<CRC><cr>: Query output load energy of year
QLMyyyymm<CRC><cr>: Query output load energy of monthQLMyyyymm<CRC><cr>: Query output load energy of month
QLDyyyymmdd<CRC><cr>: Query output load energy of dayQLDyyyymmdd<CRC><cr>: Query output load energy of day

this video explain lot more
https://www.youtube.com/watch?v=dlhsc_KjIM4&list=PLDW1zKN_tjaJDp-Ukz8KApchBgA60AFC_

dj-nitehawk, 2022-07-25:
try https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/releases/tag/v1.4.2
last one ;-)

Saentist, 2022-07-25:
simple way to debug calling string
```bash
cat /dev/hidraw2

(000.0 00.0 000.0 00.0 0000 0000 000 223 00.10 000 000 0036 00.0 228.2 00.00 00000 00000000 00 00 00000 010kX
(NAKss 00.0 000.0 00.0 0000 0000 000 223 00.10 000 000 0036 00.0 228.2 00.00 00000 00000000 00 00 00000 010kX
(NAKss

```