dj-nitehawk/Hybrid-Inverter-Monitor #6: Monitor from local network

udeshjr, 2022-08-01:
I couldn't access server from local ip. localhost works fine. When scanned from nmap (windows) "80/tcp   open  http          Kestrel". But browser end up with "This site can’t be reached192.168.8.201 refused to connect.". Slimier old post were found in following two links. 
https://stackoverflow.com/questions/34212765/how-do-i-get-the-kestrel-web-server-to-listen-to-non-localhost-requests
https://stackoverflow.com/questions/33975949/asp-net-5-kestrel-connect-within-lan/34092934#34092934

Please consider for future developments.

dj-nitehawk, 2022-08-01:
it's already configured to listen on any IP address, even non local ones: 

https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/blob/dd10a048991a35c301a8eb6fb92054ffc487fe1e/src/Server/Program.cs#L16

most likely your firewall is blocking port 80. it could be your Linux device or router doing the blocking.

app/server is working properly for me and all others as expected.

dj-nitehawk, 2022-08-01:
also you should be using http:// and not https://

closed 2022-08-01 by dj-nitehawk

Saentist, 2022-08-01:
> also you should be using http:// and not https://

If someone want to use HTTPS
then redirect :80 to :443 with 
https://www.stunnel.org

udeshjr, 2022-08-02:
what about  Hybrid-Inverter-Monitor/src/Server/Properties/launchSettings.json Line 8
"applicationUrl": "http://0.0.0.0::80;http://0.0.0.0::443", //I aleady edited the line

I will try again. When scanned from nmap (windows) from local network "80/tcp open http Kestrel".  I thought nothing blocks port 80.

dj-nitehawk, 2022-08-02:
launchSettings.json only applies during development in visual studio