dj-nitehawk/Hybrid-Inverter-Monitor #7: Connection timed out

sanatho, 2022-10-17:
When I execute InverterMon.Server I get this error 
[image]
ice.CommandExecutor[0]
OS: raspian os lite (64 bit)
Device: Raspberry pi 3b+

dj-nitehawk, 2022-10-17:
execute `lsusb` in your terminal.
see if there's a device like the following:
```cs
Bus 005 Device 002: ID 0665:5161 Cypress Semiconductor USB to Serial
```
if a device does exist, do the following and determine where the hidraw device is mounted:
```cs
ls /dev/hid*
```
if the hidraw number is not 0, you gotta update the `appsettings.json` file with the correct path to the device.

sanatho, 2022-10-17:
[image]
I see this device

sanatho, 2022-10-17:
[image]

when i execute the server for the first time i see the info for 1 second and after  i get the error

dj-nitehawk, 2022-10-17:
when connection breaks, does the `/dev/hidraw0` device also disappear from the os?
could be a loose usb connection/port issue.
does it work correctly for watchpower app?

sanatho, 2022-10-17:
I have never encountered any problems in watchpower app

dj-nitehawk, 2022-10-17:
and what is the brand/model of your inverter?
does it support the voltronic protocol?

sanatho, 2022-10-17:
this appear when i execute the server for the first time
[image]
the device doesn't disappear
[image]
the inverter is inxii 3000w (i think that support voltronic protocol because i see info the first time)

dj-nitehawk, 2022-10-17:
are you connecting to the raspi via wifi? if so, can you try connecting with ethernet instead?
if still no after trying that, i'm afraid there's no way for me to debug the issue without having physical access to your inverter.
unless you know some C# and willing to do the debugging yourself using the source code.

sanatho, 2022-10-17:
I am connected via wifi and i and i can't connect via ethernet. maybe i could try to understand c#

dj-nitehawk, 2022-10-17:
also try holding the usb port in a fixed position on the inverter without letting go. those chinese clone inverters sometimes have loose connection ports. i have no idea why the connection breaks after initial success.

dj-nitehawk, 2022-10-17:
also i've had similar experience with raspi and built-in wifi. after using a wifi dongle, the situation became better.

sanatho, 2022-10-17:
the strange thing is that the original program does not give any problems

dj-nitehawk, 2022-10-17:
yes strange indeed.

sanatho, 2022-10-17:
Thanks for your suggestions

Saentist, 2022-10-17:
@sanatho 
See if DMESG show some strange messages when this happens.

My suggestion is, that there is electrical grounding problem.

dj-nitehawk, 2022-10-17:
@Saentist can a grounding issue cause interference with serial data communication?

sanatho, 2022-10-17:
Could low power supply be a problem?

dj-nitehawk, 2022-10-17:
insufficient power may also be a cause. i've had both wifi and usb connection dropouts with raspi b3+
switched to an orange pi 4 lts and been problem free since ;-)

closed 2023-01-06 by dj-nitehawk