dj-nitehawk/Hybrid-Inverter-Monitor #8: execution error: Format_InvalidString

eloquentvogon, 2022-11-09:
Hello I am using a MPP solar LVX6048WP connected to a raspberry pi running this software

things seem to connect ok but I repeatedly get

`./InverterMon.Server info: FastEndpoints.StartupTimer[0] Registered 6 endpoints in 2326 milliseconds. info: InverterMon.Server.InverterService.CommandExecutor[0] inverter connected! info: Microsoft.Hosting.Lifetime[14] Now listening on: http://0.0.0.0:80 info: Microsoft.Hosting.Lifetime[0] Application started. Press Ctrl+C to shut down. info: Microsoft.Hosting.Lifetime[0] Hosting environment: Production info: Microsoft.Hosting.Lifetime[0] Content root path: /home/pi/solar/ fail: InverterMon.Server.InverterService.CommandExecutor[0] execution error: Format_InvalidString fail: InverterMon.Server.InverterService.CommandExecutor[0] execution error: Format_InvalidString fail: InverterMon.Server.InverterService.CommandExecutor[0] execution error: Format_InvalidString fail: InverterMon.Server.InverterService.CommandExecutor[0] execution error: Format_InvalidString fail: InverterMon.Server.InverterService.CommandExecutor[0] execution error: Format_InvalidString fail: InverterMon.Server.InverterService.CommandExecutor[0] execution error: Format_InvalidString fail: InverterMon.Server.InverterService.CommandExecutor[0] execution error: Format_InvalidString fail: InverterMon.Server.InverterService.CommandExecutor[0] execution error: Format_InvalidString fail: InverterMon.Server.InverterService.CommandExecutor[0] execution error: Format_InvalidString fail: InverterMon.Server.InverterService.CommandExecutor[0] execution error: Format_InvalidString fail: InverterMon.Server.InverterService.CommandExecutor[0] execution error: Format_InvalidString fail: InverterMon.Server.InverterService.CommandExecutor[0] execution error: Format_InvalidString ^Cinfo: Microsoft.Hosting.Lifetime[0] Application is shutting down... fail: InverterMon.Server.InverterService.CommandExecutor[0] execution error: Format_InvalidString`

Thanks

dj-nitehawk, 2022-11-10:
give this a go: https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/issues/7#issuecomment-1280887130

eloquentvogon, 2022-11-10:
thanks will take a look later on, appreciate it

eloquentvogon, 2022-11-11:
thanks I tried that and same result, I do no think it is a connection issue, I think these new inverters are spitting out newer strings which are longer and not matching the expected output. 

I have asked MPP what they are outputting, it will be interesting to see what they comeback with

eloquentvogon, 2022-11-11:
I was playing with mpp-solar and finally got some results, I think it is running PI17 

`Command: PS - Device Power Status
------------------------------------------------------------
Parameter                       Value           Unit
solar_input_power_1             41              W
solar_input_power_2             0               W
battery_power                                   W
ac_input_active_power_r         845             W
ac_input_active_power_s         564             W
ac_input_active_power_t         0               W
ac_input_total_active_power     1409            W
ac_output_active_power_r        838             W
ac_output_active_power_s        557             W
ac_output_active_power_t        0               W
ac_output_total_active_power    1395            W
ac_output_apparent_power_r      1219            VA
ac_output_apparent_power_s      559             VA
ac_output_apparent_power_t      0               VA
ac_output_total_apparent_power  1778            VA
ac_output_power_percentage      40              %
ac_output_connect_status        Connected
solar_input_1_work_status       Working
solar_input_2_work_status       Idle
battery_power_direction         Charging
dc/ac_power_direction           AC to DC
line_power_direction            Input
`

closed 2023-01-06 by dj-nitehawk