dj-nitehawk/Hybrid-Inverter-Monitor #10: Request/ Clarification : Get the inverter running mode

mihiHarshana, 2022-12-25:
I am not much into c#,  appriciate if there is any way I cant get Inverter Current mode. ie, if it is currently on 'By Pass mode' 
as per the protocol guilde, Computer: QMOD<CRC><cr> should do this. but what I don't understand here is, why it does not work ? 
any config to be changed to get that ?  
Appreciate If someone experienced can provide me a guide here. 
Thank you

dj-nitehawk, 2022-12-25:
what's your inverter model?
my sako 3.5kw inverter does not report "bypass mode".
if i remember correctly it only reports "battery mode", which can't be used to determine if the inverter is in bypass mode or not.

mihiHarshana, 2022-12-25:
Sako Sun on pro 5.5KW, I am not yet sure if it reports or not. 
I would try to change the Status that you have used 'api/status' try to get the mode and give you and update.
I am stuck on adding a new file and getting the values like Status you have done.

mihiHarshana, 2022-12-25:
I did something like this., Charging the current status and remove the rest of the code, changed to "QMOD"
I got the following results
in By Pass mode Inverter sends "L"
when in Battery mode Inverter sends "B"

Still my question is I need to add a new file as GetMode, and when I do  as you have done for GetStatus. it does not work at all.

Hope you are clear on what I am trying. Sorry for troubling you.

dj-nitehawk, 2022-12-25:
how often do you wanna query the mode? 
once in a while in a one-off fashion or do you wanna query it continuously?
if continuously, need to modify the `StatusRetriever` background service to do the polling frequently.
if once in a while, it would be easier to create a new endpoint to do the querying on request.
let me know what you want, and i'll see what can be done when i have some free time.

mihiHarshana, 2022-12-25:
I would prefer it is happening often so I know if the inverter is in what state.
Thanks