dj-nitehawk/Hybrid-Inverter-Monitor #11: Unable to read settings in a timely manner!

JeanEngelbrecht01, 2023-01-05:
Hi,

When calling api/settings/get-charge-ampere-values

I get the following response

{"statusCode":400,"message":"One or more errors occured!","errors":{"GeneralErrors":["Unable to read settings in a timely manner!"]}}

I have a Mecer 3Kva inverter, I am not sure if there is an error with getting the specific information but the api/status call seems to work.

I believe this because there is no specific error handling the whole site's JS crashes as a first point.

secondly, I am quite familiar with c# and would like to debug, can I connect my pc straight to the inverter to debug?

dj-nitehawk, 2023-01-05:
you will only be able to connect the inverter directly to you computer if your OS is linux based and it automatically mounts the inverter as an HID device. if you're on windows, it won't work. alternative would be to compile the app with some logging/debug info and see where the response from the inverter is different (or non-responsive) than the norm.

JeanEngelbrecht01, 2023-01-05:
Thank you for the feedback.

I have a MacBook as well, would VS for Mac work to debug?

dj-nitehawk, 2023-01-06:
if a device appears under `/dev/hidraw0` when you plug in the inverter directly to the macbook, then you should be good to go.

closed 2023-01-06 by dj-nitehawk