dj-nitehawk/Hybrid-Inverter-Monitor #12: Raspberry pi zero w problem

gottmoz, 2023-01-05:
Hello i have downloaded all files to /inverter library in root and when trying to start with command ./InverterMon.Server i 
get a fault illegal instruction!? Does anybody else have this problem? 
The Zero does not seem to find a Hidraw0 but a USB0...
I Used the normal Linux arm release..

Saentist, 2023-01-05:
Do you try to use it (USB0)?
I have success not to use Hidraw interface.

gottmoz, 2023-01-06:
Yes, i tried using USB
But i can’t even get the software to start, when i try to start it says illegal instruction…

dj-nitehawk, 2023-01-06:
rpi zero w is an arm v6 32bit device right?
if so, i believe the [linux-arm.zip](https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/releases/download/v1.11/linux-arm.zip) binaries should work.

is that the one you tried or something else?

dj-nitehawk, 2023-01-06:
also give https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/issues/7#issuecomment-1280887130 a try

closed 2023-01-17 by dj-nitehawk