dj-nitehawk/Hybrid-Inverter-Monitor #17: Exit code - not starting

Redzone003, 2023-03-25:
Hy, i have a litle problem - after i type the status comand, it shows me -> main pid:2410 (code=exited, status=200/CHDIR). What seems to be the problem ? I'm using Ubuntu 20.4.

dj-nitehawk, 2023-03-25:
what do you mean by "status command"?

Saentist, 2023-03-25:
> what do you mean by "status command"?

service status I guess

@Redzone003 
post result 
fortuneteller is missing

Redzone003, 2023-03-25:
After the execution of : sudo systemctl status invertermon

Saentist, 2023-03-25:
> After the execution of : `sudo systemctl status invertermon`

Posing result from terminal help others to see what you see.

dj-nitehawk, 2023-03-25:
i'm assuming the binary craps out because you've downloaded the incorrect binary for your device's architecture.
what are you running this on? 

execute the binary manually `./InverterMon.Server` and post the output as well like @Saentist suggested.

the more detail you give us, easier it is to get an idea what's happenning.

Redzone003, 2023-03-25:
it's running on an old intel proccesor - i3/370m and i've downloaded the "linux-x64" zip file.
like you sugested, after the execution of the binary, here is the result:

info: FastEndpoints.StartupTimer[0]
      Registered 7 endpoints in 1364 milliseconds.
warn: InverterMon.Server.BatteryService.JkBms[0]
      trying to open bms port at: /dev/ttyUSB0
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      inverter connection failed: UnauthorizedAccess_IODenied_Path, /dev/hidraw0
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      inverter connection failed: UnauthorizedAccess_IODenied_Path, /dev/hidraw0
fail: InverterMon.Server.InverterService.CommandExecutor[0]
      inverter connection failed: UnauthorizedAccess_IODenied_Path, /dev/hidraw0
warn: InverterMon.Server.BatteryService.JkBms[0]

Redzone003, 2023-03-25:
I've executed the binary with sudo and now the server started, but something it's not good.When i try to enter in setting from the app, nothing happens (the page stays on loading), and meanwhile in terminal i have the following error:

fail: Microsoft.AspNetCore.Server.Kestrel[13]
      Connection id "0HMPDDQBS8M29", Request id "0HMPDDQBS8M29:0000005B": An unhandled exception was thrown by the application.
      System.Threading.Tasks.TaskCanceledException: TaskCanceledException_ctor_DefaultMessage
         at InverterMon.Server.InverterService.Commands.Command`1.WhileProcessing(CancellationToken c)
         at InverterMon.Server.Endpoints.Settings.GetChargeAmpereValues.Endpoint.HandleAsync(CancellationToken c)
         at FastEndpoints.Endpoint`2.ExecAsync(CancellationToken ct)
         at FastEndpoints.Endpoint`2.ExecAsync(CancellationToken ct)
         at Microsoft.AspNetCore.Authorization.AuthorizationMiddleware.Invoke(HttpContext context)
         at Microsoft.AspNetCore.Server.Kestrel.Core.Internal.Http.HttpProtocol.ProcessRequests[TContext](IHttpApplication`1)

Saentist, 2023-03-25:
`sudo ./InverterMon.Server`
regular user do not have full access to all devices

Redzone003, 2023-03-25:
Yes, i've done it allready...but i encountered another issue - please see in the previous post.

Saentist, 2023-03-25:
still less then needed information
What os you use?
What connection to inverter?

Redzone003, 2023-03-25:
Ubuntu 20.4, and connection through micro usb cable.

dj-nitehawk, 2023-03-26:
> InverterMon.Server.Endpoints.Settings.GetChargeAmpereValues.Endpoint.HandleAsync()

this leads me to believe the inverter didn't respond in a timely mannery when querying for the charge current values.

@Redzone003 what's your inverter make/model?

getting the live status works fine yes?
only issue is you can't load the inverter settings page yes?

Redzone003, 2023-03-26:
Live status works fine only the first time when open the app, and then never refresh.
The inverter is a mpp solar - 6048mt. 
I will try the commands you sugested and i will provide a feedback. Thank you.

Redzone003, 2023-03-26:
1.  sudo mpp-solar -p /dev/hidraw0 --getDeviceId
Command: Get Device ID - Generate a device id
--------------------------------------------------------------------------------
Parameter Value          	Unit
deviceid  PI30:055:VMIII-6000

2. sudo mpp-solar -p /dev/hidraw0 -c QMUCHGCR -I
2023-03-26 22:32:28,773:INFO:__init__:main@213: Solar Device Command Utility, version: 0.15.24
2023-03-26 22:32:28,773:INFO:__init__:main@346: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI30"
2023-03-26 22:32:28,775:INFO:__init__:get_port@97: Using hidrawio for communications
2023-03-26 22:32:28,779:INFO:__init__:main@406: Looping 1 commands
2023-03-26 22:32:28,779:INFO:__init__:main@416: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x7f37811610a0>, protocol: PI30 protocol handler for command: QMUCHGCR, tag: QMUCHGCR, outputs: screen
2023-03-26 22:32:28,780:INFO:device:run_command@50: Running command QMUCHGCR
2023-03-26 22:32:28,780:INFO:abstractprotocol:get_full_command@48: Using protocol b'PI30' with 42 commands
2023-03-26 22:32:28,780:INFO:device:run_command@78: full command b'QMUCHGCR&4\r' for command QMUCHGCR
2023-03-26 22:32:30,693:INFO:abstractprotocol:decode@217: response passed to decode: b'(002 010 020 030 040 050 060 070 080 090 100\\F\r'
2023-03-26 22:32:30,694:INFO:abstractprotocol:decode@261: Processing response of type DEFAULT
2023-03-26 22:32:30,694:INFO:abstractprotocol:decode@272: Processing DEFAULT type responses
2023-03-26 22:32:30,694:INFO:device:run_command@113: Decoded response {'_command': 'QMUCHGCR', '_command_description': 'Max Utility Charging Current Options inquiry', 'raw_response': ['(002 010 020 030 040 050 060 070 080 090 100\\F\r', ''], 'Max Utility Charging Current': ['002', 'A'], 'Unknown value in response 1': ['010', ''], 'Unknown value in response 2': ['020', ''], 'Unknown value in response 3': ['030', ''], 'Unknown value in response 4': ['040', ''], 'Unknown value in response 5': ['050', ''], 'Unknown value in response 6': ['060', ''], 'Unknown value in response 7': ['070', ''], 'Unknown value in response 8': ['080', ''], 'Unknown value in response 9': ['090', ''], 'Unknown value in response 10': ['100', '']}
2023-03-26 22:32:30,694:INFO:__init__:get_output@40: attempting to create output processor: screen
2023-03-26 22:32:30,696:INFO:screen:output@21: Using output processor: screen
Command: QMUCHGCR - Max Utility Charging Current Options inquiry
--------------------------------------------------------------------------------
Parameter                    Value          	Unit
max_utility_charging_current 002            	A   
unknown_value_in_response_1  010            	    
unknown_value_in_response_2  020            	    
unknown_value_in_response_3  030            	    
unknown_value_in_response_4  040            	    
unknown_value_in_response_5  050            	    
unknown_value_in_response_6  060            	    
unknown_value_in_response_7  070            	    
unknown_value_in_response_8  080            	    
unknown_value_in_response_9  090            	    
unknown_value_in_response_10 100

Saentist, 2023-03-26:
`sudo mpp-solar -p /dev/hidraw0 -c QMUCHGCR -P`

Redzone003, 2023-03-26:
sudo mpp-solar -p /dev/hidraw0 -c QMUCHGCR -P
Command: protocols help - List available protocol modules
--------------------------------------------------------------------------------
Parameter  Value          	Unit
daly       Daly BMS protocol handler	    
daly40     Daly40 BMS protocol handler	    
jk02       JK02 - JKBMS BLE 2 byte data communication protocol handler	    
jk04       JK04 - JKBMS BLE 4 byte data communication protocol handler	    
jk232      JKBMS RS232 serial communication protocol handler	    
jk485      JKBMS RS485 serial communication protocol handler	    
jkv11      JKBMS BLE communication protocol handler software v11.x	    
pi16       PI16 protocol handler	    
pi17       PI17 protocol handler	    
pi17infini PI17INFINI protocol handler	    
pi18       PI18 protocol handler	    
pi18sv     PI18 protocol handler for Easun SV inverters	    
pi30       PI30 protocol handler	    
pi30max    PI30 protocol handler for LV6048MAX and similar inverters	    
pi30mst    PI30 protocol handler for PIP4048MST with 3 PV and similar inverters	    
pi30revo   PI30 protocol handler for REVO and similar inverters	    
pi41       PI41 protocol handler	    
ved        VED protocol handler for Victron direct SmartShunts

Saentist, 2023-03-26:
> ## sudo mpp-solar -p /dev/hidraw0 -c QMUCHGCR -P
> Command: protocols help - List available protocol modules
> Parameter Value Unit daly Daly BMS protocol handler daly40 Daly40 BMS protocol handler jk02 [...]

logic is to put protocol
pi30max for example

dj-nitehawk, 2023-03-27:
> Live status works fine only the first time when open the app, and then never refresh.

@Saentist does this sound like a connection dropping issue? probably related to usb/power/interference?

@Redzone003 can you post what the terminal shows after executing the `InverterMon.Server`?
just wanna see if there's any errors related to usb connection being dropped.

Redzone003, 2023-03-27:
I've attached 2 pictures, one with the app in the settings menu and another one, with the terminal, after executing the setting button from the app, or after i hit refresh on the page.
[image]
[image]

Saentist, 2023-03-27:
In second terminal window 
`sudo dmesg -wHT`
when 
`sudo ./InverterMon.Server`
is started.

dj-nitehawk, 2023-03-28:
can't see anything obviously wrong in the log.
so i'm kinda stumped as to why it shows the live status once and stops.
most likely some sort of communication break with the inverter.
this kind of issue is really difficult to debug unless i have access to the particular inverter+computer combo in question.
what i can suggest is to try it on a different linux computer if you have one handy.
i used to get intermittent connection issues with a raspberry-pi i had and it got sorted by itself when i switched to using an orange-pi.

i'll close this issue since there's not much else i can offer.

you guys can keep chatting if you like and see if something can be sorted out.

closed 2023-03-28 by dj-nitehawk

Redzone003, 2023-03-28:
I've tried with another computer and this time i installed linux mx 21.2 - the same result. What configuration do you use, guys ? What os ?

Saentist, 2023-03-28:
Try with rg45>rs232 cable with come with inverter

I use Ubuntu 20.04 on old Asus AiO.

WatchPower also work under Linux [LINK](http://www.mppsolar.com/manual/Watchpower%20%28off-grid%29/WatchPower1.16SP7/installWatchPower_Linux_64bit.bin)