dj-nitehawk/Hybrid-Inverter-Monitor #19: Settings "An error has occurred... Reload"

iboboc, 2023-04-02:
Hi there,

I have a EASUN inverter SMP-5k, 5kw hybrid Chinese. The code is working (latest release - but was similar on older ones as well), on live stats and on PV Generation I can see the data from /hidraw1, but if I go in Settings page, is not loading and I have an error: "An error has occurred... Reload". 

If I check the service status I can see only this:

Apr 02 21:24:01 linux InverterMon.Server[1011]: info: Microsoft.Hosting.Lifetime[14]
Apr 02 21:24:01 linux InverterMon.Server[1011]:       Now listening on: http://0.0.0.0:80
Apr 02 21:24:01 linux InverterMon.Server[1011]: info: Microsoft.Hosting.Lifetime[0]
Apr 02 21:24:01 linux InverterMon.Server[1011]:       Application started. Press Ctrl+C to shut down.
Apr 02 21:24:01 linux InverterMon.Server[1011]: info: Microsoft.Hosting.Lifetime[0]
Apr 02 21:24:01 linux InverterMon.Server[1011]:       Hosting environment: Production
Apr 02 21:24:01 linux InverterMon.Server[1011]: info: Microsoft.Hosting.Lifetime[0]
Apr 02 21:24:01 linux InverterMon.Server[1011]:       Content root path: /inverter
Apr 02 21:24:01 linux InverterMon.Server[1011]: fail: InverterMon.Server.InverterService.CommandExecutor[0]
Apr 02 21:24:01 linux InverterMon.Server[1011]:       execution error: Format_InvalidStringWithValue, NAKs

Any idea what is wrong?

Saentist, 2023-04-02:
Inverter commands support is not universal.
[link](https://github.com/softwarecrash/Solar2MQTT/files/8682584/Communication.protocol.powland.English-Auto.Translated.pdf)

iboboc, 2023-04-03:
Sure, thanks for quick reply. It is working with WatchPower, which works with many inverters, so I assumed it is the same protocol and should work out of the box.
I'll try to investigate it. Thanks!

dj-nitehawk, 2023-04-03:
> Apr 02 21:24:01 linux InverterMon.Server[1011]: execution error: Format_InvalidStringWithValue, NAKs

this line means, the inverter didn't accept a command sent from the program. most likely the command needed to get charge current values which are needed to be displayed in the dropdown list for changing charging currents.

unfortunately, there's no way for me to add support for your specific inverter without actually having one to test.

closed 2023-04-03 by dj-nitehawk

Saentist, 2023-04-03:
> unfortunately, there's no way for me to add support for your specific inverter without actually having one to test.

QID+QMN+QGMN = device set of supported commands.

iboboc, 2023-04-03:
If you have time/curiosity to look into this, I can facilitate you access to the linux machine where the inverter is connected (i can give you a user and ssh access. This way you can test/try anything.

dj-nitehawk, 2023-04-15:
@iboboc try the latest version. if it still gives trouble, update the `appsettings.json` file to read:
```json
"TroubleMode" : "yes"
```
and run the app.

in "trouble mode", some commands such as getting charging ampere values or setting some inverter settings might not work and lock up the data retrieval. if that happens, restarting the app should get it going again.