dj-nitehawk/Hybrid-Inverter-Monitor #23: Serenum Axpert Inverter - data freeze using USB

geofreybingachie, 2023-07-10:
I have tried using your software but I have a challenge. Every time i try to query inverter settings the device communication freezes and the values on the Live stats tab change to zero . Even when I leave the system idle a few hours later I find the inverter part frozen with some values or all values are zero . I have tried your software on raspberry pi 3b running Debian and also trying the same with an x86 pc running latest Ubuntu server Lts.Help me identify what the problem might be.
FYI i even tried a different USB cable .

dj-nitehawk, 2023-07-11:
can you try the following:

1. install mppsolar from jblanch
2. update the following fields in `appsettings.json` file
https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/blob/6ea7af8b5c78c423c7075c0ff4b9d9cda93aebb7/src/Server/appsettings.json#L6-L7
set `"TroubleMode" : "yes"` and set the correct path for the mppsolar binary if it's different than what's shown.
3. restart the app

if that doesn't help, i have no idea how to troubleshoot inverters that i don't have physical access to :-(

closed 2023-07-11 by dj-nitehawk

geofreybingachie, 2023-07-11:
My MPP solar path was wrong, i have made the change hopefully it will work flawlessly now .I will update you in the next 24 hours . Thank you.

Saentist, 2023-07-12:
I don't think cable have any problem, 
most probably pc or port go to powersave mode.

geofreybingachie, 2023-07-15:
> I don't think cable have any problem, most probably pc or port go to powersave mode.

I have tried with three different installations ( a desktop pc, a USFF pc, and raspberry pi 3b ) . I have also noted the freeze can happen if I try to query the inverter settings. I even bought a new USB cable to try a test if the cable is the problem.