dj-nitehawk/Hybrid-Inverter-Monitor #26: website remains empty

bab200, 2023-10-07:
Hi - I just installed my Voltronic Power VMIII5000-48. All runs well. Now I wanted to have a good way to monitor the solar system and found this script.

I got it up and running - but when opening the web browser, the website is empty. on my raspberry, I am getting this:

any idea?

warn: InverterMon.Server.BatteryService.JkBms[0]
      trying to open bms port at: /dev/ttyUSB0
info: FastEndpoints.StartupTimer[0]
      Registered 7 endpoints in 3206 milliseconds.
info: InverterMon.Server.InverterService.CommandExecutor[0]
      connecting to the inverter...
info: InverterMon.Server.InverterService.CommandExecutor[0]
      connected to inverter at: [/dev/hidraw0]
info: Microsoft.Hosting.Lifetime[14]
      Now listening on: http://0.0.0.0:80
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: /home/pi
info: Microsoft.AspNetCore.Hosting.Diagnostics[1]
      Request starting HTTP/1.1 GET http://192.168.1.48/ - -
info: Microsoft.AspNetCore.Routing.EndpointMiddleware[0]
      Executing endpoint 'Fallback {*path:nonfile}'
info: Microsoft.AspNetCore.Routing.EndpointMiddleware[1]
      Executed endpoint 'Fallback {*path:nonfile}'
info: Microsoft.AspNetCore.Hosting.Diagnostics[2]
      Request finished HTTP/1.1 GET http://192.168.1.48/ - - - 404 0 - 1101.8171ms
warn: InverterMon.Server.BatteryService.JkBms[0]
      trying to open bms port at: /dev/ttyUSB0

dj-nitehawk, 2023-10-27:
> Request finished HTTP/1.1 GET http://192.168.1.48/ - - - 404 0 - 1101.8171ms

that shouldn't be happenning. maybe some files in the `wwwroot` folder got deleted/corrupted. or a lack of read permissions. no idea really. only thing i can recommend is to try again from scratch.

closed 2023-10-27 by dj-nitehawk

Saentist, 2023-10-28:
Change Default port