dj-nitehawk/Hybrid-Inverter-Monitor #28: Request for Improvement - Can We improve to see the monthly generation, Year generation like wise.

mihiHarshana, 2024-02-04:


dj-nitehawk, 2024-02-04:
sorry brother...
don't really have the time these days.
will gladly accept a PR from the community.

orcuncolakoglu, 2024-06-06:
Hello nitehawk. This is something great that I use at home. How can I open InverterMon.db file so I can prepare something for generation graphs?

dj-nitehawk, 2024-06-06:
@orcuncolakoglu that's [litedb](https://www.litedb.org/) data file. there's a lite db studio [gui app](https://github.com/mbdavid/LiteDB.Studio) you can use to browse the data. alternative is to write your own .net/c# tool using the litedb library that will read/export the pv generation table/collection.

orcuncolakoglu, 2024-06-06:
@dj-nitehawk I believe there is a password in the db? As I it asks for it...

dj-nitehawk, 2024-06-06:
there's no password. the app just opens it via file name:
https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/blob/6ea7af8b5c78c423c7075c0ff4b9d9cda93aebb7/src/Server/Persistance/Database.cs#L22

just tried it on my own database, works fine: https://streamable.com/eo033y