dj-nitehawk/Hybrid-Inverter-Monitor #30: Request for VenusOS/Home assistant compatibilty

FrankieHAi, 2024-03-03:
Hello,
Great project! I'm new to this so please excuse my lack of knowledge.
Is it possible to use this to add MPP Solar inverter and JKBMS compatibility to Victron VenusOS and/or Home Assistant running on RPI? This would be very useful if possible!

Thanks Paul

dj-nitehawk, 2024-03-03:
this was made to work standalone simply by plugging in an mppsolar inverter and a jk bms.
there is no built-in integration with any other service or system such as victron/home assistant/grafana/etc.
somebody with knowledge of those systems and some C# might be able to just fork this repo and add those integrations since the logic for reading data from inverter/bms is already there.

since i do not have/use any of those systems, i can't make that myself. nor do i have the desire to 😜

if someone wants to have crack at it, i can only answer any specific questions regarding the code i've already written.

closed 2024-03-03 by dj-nitehawk

FrankieHAi, 2024-03-03:
> this was made to work standalone simply by plugging in an mppsolar inverter and a jk bms. there is no built-in integration with any other service or system such as victron/home [...]

Thanks for your reply!
That's beyond my scope, though someone else might run with it