dj-nitehawk/Hybrid-Inverter-Monitor #31: sumry hybrid inverter is supported?

hoangkhacchung, 2024-04-11:
I try connecting to sumry hybrid inverter but it is not working.  sumry hybrid inverter is supported? Thanks,

dj-nitehawk, 2024-04-11:
is it a voltronic clone inverter?
does it work with the watchpower app via usb cable?
if so, it should theoretically work.
can't offer any more help than that as i'm not familiar with sumry brand.

closed 2024-04-11 by dj-nitehawk

Saentist, 2024-04-11:
@dj-nitehawk 
enable discussion module in repo for "question's"

dj-nitehawk, 2024-04-11:
@Saentist got too many things to manage already. it's ok if ppl wanna ask questions in gh issues ;-)

Saentist, 2024-04-11:
@dj-nitehawk discussion is for that ;)
can be made section "supported/tested inverters"
easy for all, after some time issue will be closed and hard to find needed information.