dj-nitehawk/Hybrid-Inverter-Monitor #32: How to compile

gnuorder, 2024-05-16:
I'm trying to tweak this to match my inverter. How do I compile the server portion? I'm not familiar with c# or DOTNET. I'm using dotnet build -o out InverterMon.Server.csproj but I need all the support dlls for it to run so I test run it out of the out dir and the web page doesn't fully load.

[InverterMon.log](https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/files/15340457/InverterMon.log)

dj-nitehawk, 2024-05-17:
try dotnet publish with the correct system architecture and the self contained flag and well as done in the gh workflow: 

https://github.com/dj-nitehawk/Hybrid-Inverter-Monitor/blob/6ea7af8b5c78c423c7075c0ff4b9d9cda93aebb7/.github/workflows/create-release.yaml#L34 

it should just spit out an executable binary.

closed 2024-05-17 by dj-nitehawk

gnuorder, 2024-05-17:
Thank you. That works. The temp is in a different spot with my inverter and I couldn't find PV watts at all so I calculated it from PV volts and amps.